OLIMEX/ESP32-POE-ISO #4: Ethernet Connection as esp sleep wake source

lucaelin, 2021-04-10:
Is it possible to use the ethernet connection as a wake source for the esp32 deep sleep? 
I couldn't find any documentation for this and the ethernet RX and TX are directly connected to the LAN8710A with no way to detect the switches auto-negotiation signals using the esp. AFAIK using the external power pin would work only if the switch supplies PoE...
Passing wires from the jacks RX lines to the GPIOs would work and not violate the galvanic isolation, right? Or does the LAN8710A provide another way to detect a connection without being powered?
Thanks in advance!

DanKoloff, 2021-04-12:
Probably any GPIO can be used as external source for wake up, but we haven't tested it. Please check what is written here:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html

Look for "ext0" and "ext1". You can probably set the RTC IO pins to the pins of the Ethernet, so they won't go into deep sleep.

LAN8710A has such pin but we haven't routed it, so it is unavailable. If you look into LAN8710A's datasheet look for "nINT"

closed 2021-12-07 by DanKoloff

lucaelin, 2021-12-07:
I am sorry I didn't get back to this...
I approached this issue several times looking deep into the reference that you provided, but I still was not able to solve this. From my understanding, once that esp is in sleep mode, the ethernet looses its clock to transfer any data to the esp. I've tried multiple things to get around it, but no solution would ultimately wake the esp. In short I'm still stuck with this issue, but I am also a little out of my comfort zone here.

DanKoloff, 2021-12-08:
By default this is the expected outcome since Ethernet and WIFI are disabled in all power down modes. That is why you need to "hack" other ways to wake up, aka use signals arriving at Ethernet connector and route them to ext0 and ext2 that can be used as wake up source. 

My advice is to first place a button on ext0 or ext1 pin then test it as wake up source, then figure how to do it with the Ethernet connector.

The wake up sources can be:

-Timer
-Touch pad
-External wakeup(ext0 & ext1)