OLIMEX/ESP32-POE-ISO #6: Can't find 3D model of ESP32-POE-ISO

mattbue, 2021-12-07:
Hi,
I'm searching for a 3D model of the ESP32-POE-ISO.
Unfortunatelly in the 3d folders i can only find 3D models of the parts that are mounted on the pcb.

Do i miss something?

Thanks

DanKoloff, 2021-12-07:
Download the whole repo, start KiCAD, load the project in the folder of the proper hardware revision in KiCAD (it is open source and free), start the PCB part of the design and select view -> 3D. 

Notice that the 3D view and model have illustrative purposes only! Do not trust the 3D model blindly, the heights of components might be inaccurate. This is because we don't need the heights for manufacturing and some components heights are not exact.

mattbue, 2021-12-07:
Thank you for your quick response.
I'm not familiar with KiCAD.
Since for mechanical construction I'm using SolidWorks was searching for a STEP-file containing the complete model.
Is there any chance to get such a STEP-file?

DanKoloff, 2021-12-07:
I generated a STEP file from KiCAD I check if it loads in some free STEP viewer and it seems fine. I have no experience with SolidWorks so I can't verify the export. You can find it inside the archive here: 

[ESP32-PoE_Rev_G1_STEP.zip](https://github.com/OLIMEX/ESP32-POE-ISO/files/7666826/ESP32-PoE_Rev_G1_STEP.zip)

My advice is to install KiCAD, you might need it to analyze the electrical design or trace routes and wires. It is one of the big perks of ESP32-POE-ISO's design - an open-source hardware board, made with free open-source software CAD software. KiCAD can also be used to measure distances in the PCB view, there is a ruler tool.

mattbue, 2021-12-07:
Thank you!
Importing into SolidWorks worked and the board are looking great!

DanKoloff, 2021-12-07:
Great news, glad to help! Remember what I wrote previously about height and double verify empirically before you commit final design based on the this STEP. The height of components was never double checked and it serves only illustrative purposes.

mattbue, 2021-12-07:
Sure. 
For me the 3D model is just a very good base for designing a case for my application.
Before design freeze all relevant dimensions will be double checked.

mattbue, 2021-12-08:
I've recognized that you have provided the 3D model of ESP32-PoE.
Can you please also provice the 3D model of ESP32-PoE-ISO?
Thanks

DanKoloff, 2021-12-08:
My bad, you can find ISO one here:

[ESP32-PoE-ISO_Rev_G1.zip](https://github.com/OLIMEX/ESP32-POE-ISO/files/7676229/ESP32-PoE-ISO_Rev_G1.zip)

mattbue, 2021-12-08:
Nothing to worry about!
Thanks

closed 2022-10-31 by TsvetanUsunov