OLIMEX/ESP32-POE-ISO #7: DAC pins or analog outputs

omerll, 2021-12-16:
Hey mate, hope you doing well.
Do you know if the ESP32-POE-ISO has DAC outputs as the standard ESP32 (GPIO25 and GPIO26)? 
I cant find them on the pin map pdf doc, and also point which pins work stable for ADC voltage measuring??

In advance TY
Have a great day

DanKoloff, 2021-12-16:
GPIO25 and GPIO26 are used for the Ethernet. If you want to use them you need to pick a board without Ethernet.

There are a lot of ADC pins available.

closed 2021-12-16 by DanKoloff

omerll, 2021-12-16:
I need to use this board (ESP32-POE-ISO) have 4 currently. I will use MCP4725 DAC via i2c

TY

DanKoloff, 2021-12-17:
> I need to use this board (ESP32-POE-ISO) have 4 currently. I will use MCP4725 DAC via i2c

Good solution. It is generally good idea to use external hardware if you need analog pins with ESP32. Maybe also check UEXT extension boards like MOD-IO2 or MOD-IO.  They can be plugged into the UEXT of ESP32-POE-ISO.

https://www.olimex.com/Products/Modules/IO/MOD-IO2/open-source-hardware

https://www.olimex.com/Products/Modules/IO/MOD-IO/open-source-hardware