OLIMEX/ESP32-POE-ISO #10: 3D Integrated Models

markattecusa, 2023-03-29:
@DanKoloff 

Is it possible for 3d integrated models to be published with PCB changes? Last posting seems to be G1. 

https://github.com/OLIMEX/ESP32-POE-ISO/blob/master/HARDWARE/ESP32-PoE-ISO-Rev.G1/ESP32-PoE-ISO_Rev_G1.step

I can produce it if helpful, but someone may already have it.

Thanks!

DanKoloff, 2023-04-03:
Sorry I don't understand the request. Can you explain it with more words and examples?

DanKoloff, 2023-09-15:
If you request for STEP file you can export STEP via KiCAD - you can export STEP file via KiCAD (and KiCAD is free and open-source). Last time customers reported that the export works fine without any extra steps to take. Check my exchange with a customer:

https://github.com/OLIMEX/ESP32-POE-ISO/issues/6

closed 2023-09-15 by DanKoloff