OLIMEX/ESP32-POE-ISO #12: Usage of 2 DC/DCs - why?

markusdd, 2023-06-27:


DanKoloff, 2023-06-28:
WRB4805S-3WR2 doesn't seem to have enable pin. 

Basically there is suggested hardware setup when you use TPS2375PW.

markusdd, 2023-06-28:
Understood, but still it seems to be kind of twice the hardware. And also efficiency loss because you go through 2 DC/DCs.
If the isolating DC/DCs lack an enable couldn't this be done by using a simple low-RDS_on FET t turn it off when not needed?
Also: What is the point of enable? If there is no vooltage coming out f the ethernet jack?

DanKoloff, 2023-06-28:
Enable is connected to TPS2375PW's PG pin. What is TPS2375PW's PG pin? It is explained in this document: https://www.ti.com/lit/ds/symlink/tps2375.pdf

Also twice the hardware is not always bad thing if it is four times lower the price or reduces heat over a component.

markusdd, 2023-06-28:
That is true, but the reason I am asking is because the current isolating DC/DC is not goood enough for the platfoorm I am targeting an I am looking into designing something that can drive more than 400mA at 5V.
Looking at isolating DC/DCs that manage 600mA up to even 1200mA, at 78% or even more efficiency, so I am not sure which benefit the extra DC/DC brings except more cost, area and moore external components.

DanKoloff, 2023-06-28:
I understand but unfortunately ESP32-POE-ISO design is not suitable to for such loads. Maybe consider ESP32-POE, it is not isolated design but allows double the wattage: https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware

markusdd, 2023-06-28:
I guess I have to come up with something myself because isolation is a must for my applicatioon.

closed 2023-09-15 by DanKoloff