OLIMEX/ESP32-POE-ISO #13: ESP32-POE-ISO-WROVER

pmilicevic, 2023-09-14:
Hello,

I just received subject hardware in version K, that I intend to use with ESPHome. However turns out that tis release is using LAN chip LAN8710A that is not supported by ESPHome as per:
https://esphome.io/components/ethernet.html

I tried to use LAN8720 but no success. Error message is something about powering LAN chip.

Any suggestion?

Thanks!

DanKoloff, 2023-09-15:
1) If you scroll down the link you sent me you will see there is support and configuration for Olimex ESP32-POE which uses same Ethernet chip, meaning it will work. Also there is minor difference between LAN8720 and LAN8710A, the config for LAN8720 would probably work just fine
2) A lot of customers have reported success with ESPHome and Olimex ESP32 boards in the past, which leads me to believe it would work just fine
3) It is important to notice that configurations for ESP32-POE-ISO won't work out-of-the-box with ESP32-POE-ISO-WROVER, since WROVER uses two extra pins one of which is related to the clock of the Ethernet. Remember to change the clock pin assignment for GPIO17 to GPIO0 (clk_mode: GPIO0_OUT) from. Just use the config for ESP32-POE but change clk_mode to GPIO0_OUT. This is visible in the middle of the schematic of ESP32-POE-ISO!
4) We are not affiliated with ESPHome, if there are issues with ESPHome please report to ESPHome community or team.

pmilicevic, 2023-09-16:
Many thanks for your support!

It made it working with following snippet from yaml configuration:

# Olimex ESP32-POE-ISO-WROVER Rev K
esp32:
  board:  esp32-poe
  framework:
    type: esp-idf  

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_OUT
  phy_addr: 0
  power_pin: GPIO12

closed 2023-09-16 by pmilicevic